wickerwaka/PicoROM #2: 3v3-5v support for the address bus?

giulioz, 2024-02-06:
I'm seeing that in the schematics the address bus is connected directly to the PI Pico without any level shifting. Does this means that it's only compatible with 3v3 ROM interfaces and not 5v?

wickerwaka, 2024-02-06:
I have used it exclusively with 5v systems. The RP2040 is not specced as being 5v tolerant, but it seems to handle it pretty well. I wouldn't sell it in this form but I have used it like this for hundreds of hours now without problems. Fitting two more level shifters on there would be pretty difficult and would probably require putting components on the other side of the board.

I use a buffer and level shifter on the data pins because I wanted to ensure it was driving the data lines at a high enough voltage and I also wanted to protect the data pins from 5v input while they were configured as an output, which apparently can be an issue. It also simplified the Output Enable and Chip Select logic in the firmware.

https://hackaday.com/2023/04/05/rp2040-and-5v-logic-best-friends-this-fx9000p-confirms/

giulioz, 2024-02-06:
Neat! Thank you :D

closed 2024-02-06 by giulioz

ramapcsx2, 2024-02-07:
Just wanted to add, I read somewhere that one of the designers said that the IOs are practically 5V tolerant, but they couldn't officially claim that for some minor detail .. or so :)