wickerwaka/PicoROM #4: Does it work with the rp2040 16Mb?

tspfreitas, 2024-03-20:
There is the possibility of using the RP2040 16Mbit, to have a 2MByte rom being compatible with M27c160, the GPIO pinout would have to be revised, but would it be possible? even with some modification to the stock rp2040 changing its spi flash to a 16Mb flash? I found this code very interesting, and I believe it can open doors for many projects, thank you very much for sharing my congratulations

wickerwaka, 2024-03-20:
A 16MBit ROM would not be possible. The flash memory is only used for storing the firmware, configuration data and a copy of the ROM data. The ROM data itself needs to be in RAM, which is only 256Kbytes (2MBit) so that is the maximum size. Even if there was enough RAM space or flash ROM was fast enough, the next limitation would be pin count. All of the GPIO pins are being used. Every doubling in ROM size would require an additional GPIO pin.

closed 2024-03-20 by wickerwaka

JonPrevost, 2024-05-21:
Just a thought; a high speed 3-to-8 or 4-to-16 decoder could save pins. Still limited by the internal RAM size. External SRAM would multiply the project cost.

ramapcsx2, 2024-05-22:
Also, external SRAM tends to access much slower. It really ought to be internal SRAM, and hence everyone waits for the next Pico chip.. hopefully! :)