wickerwaka/PicoROM #7: Bare Pico Pinout

hachirokumiku, 2024-08-28:
What's the Bare Pico Pinout? I am curious.

wickerwaka, 2024-08-28:
Happy to answer, but I don't understand the question.

hachirokumiku, 2024-08-28:
Ah, I see it uses an I/O Expander, thanks anyway.

hachirokumiku, 2024-08-29:
> Happy to answer, but I don't understand the question.

I expanded and rephrased my question, in short, i see you built a custom board, but could this be used with a regular pi pico?

wickerwaka, 2024-08-30:
The address pins are GPIO 0-17 and the data pins are GPIO are 22-29. The Pico Pi is 40 pins while this is intended to be used in place of 32-pin ROMS and the layout of the pins with the pico pi would not be compatible with any standard ROM pin layouts because of how the GND and power pins are placed.

dshadoff, 2024-09-28:
...Not to mention that there are several GPIOs which aren't exposed on the Pico board (GPIO 23, 24, 25, 29), so even if one created a "converter" board, and even if that converter board didn't have problems with physical dimensions, it still wouldn't be sufficient electrically.

closed 2024-12-11 by wickerwaka