wickerwaka/PicoROM #8: Feature Request: Raspberry Pi Pico 2

hobopanda, 2024-09-26:
Looking at the new Pico 2 and seen that it has 520K SRAM onboard vs 264K on the Pico 1. With that added 256K you could add support for 512x8 chips with the faster 100ns speed. In my readings so far the new pico 2 also runs faster when overclocked. with faster CPU timings might get faster then 100ns.

wickerwaka, 2025-01-27:
It's something I will investigate once the chips are more available. I'm not sure how much difference the CPU improvements will make, there are some latency differences here and there that might enable some new optimizations.

closed 2025-01-27 by wickerwaka

UffeJakobsen, 2025-01-27:
Please be aware of the current issues with the RP2350

Especially errata E9 causes much concern for others - but a guess that the buffers (may) save you...

RP2350 Errata E9
https://hackaday.com/2024/09/04/the-worsening-raspberry-pi-rp2350-e9-erratum-situation/
https://hackaday.com/2024/09/20/raspberry-pi-rp2350-e9-erratum-redefined-as-input-mode-leakage-current/
https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf#page=1350