wickerwaka/PicoROM #9: VSYS on the schematic ?

UffeJakobsen, 2024-11-07:
Hi,

Interesting project - thanks for sharing.

One question - the schematic does not show any handling of VSYS in respect produce 3V3 (as far a I can see)

Does that mean that USB needs to be connected in order to feed the Pico ?

wickerwaka, 2024-11-08:
Yes. In the current designs you need to power it separately through USB. I did this for a few reasons:

- My primary use case is development and iteration, so I always have then hooked up to USB
- While you can store ROM images in flash memory it has to be copied to RAM at startup. I don't know how long this process takes and whether it would be too long for the startup times of most devices.
- I didn't know what the current draw of the PicoROM would be and didn't build it with any concern for that so I was worried about placing any additional load on systems

The current draw turns out to be pretty low (30-50mA). I haven't measure how long the flash copy will take but now that the PicoROM can control an external reset line it, startup time may not be as much of an issue. So for a future revision I may add support for power it without a USB connection.

ramapcsx2, 2024-11-08:
Btw, look at main regulator undervolting for power saving, as well as putting either or both CPU cores to true sleep in intervals.
On ESP32 sleeping the cores is an RTOS operation, where a timer wakes it up some micros later. It conserves a good 50% power.

UffeJakobsen, 2024-11-08:
> Btw, look at main regulator undervolting for power saving, as well as putting either or both CPU cores to true sleep in intervals. On ESP32 sleeping the cores is an RTOS operation, where a timer wakes it up some micros later. It conserves a good 50% power.

@ramapcsx2 
That sounds like it is way too slow - this product is essentially a ROM/EPROM emulator 
Even when targeting 70'iies, 80'ies it needs to respond in speeds around ~100-200(-300) nano seconds

UffeJakobsen, 2024-11-08:
Hi @wickerwaka 

> Yes. In the current designs you need to power it separately through USB. I did this for a few reasons:
> 

thanks for clarifying (I thought that I was going blind - looking for a connection between VSYS and VBUS :-D )

> 
> The current draw turns out to be pretty low (30-50mA). I haven't measure how long the flash copy will take but now that the PicoROM can control an external reset line it, startup time may not be as much of an issue. So for a future revision I may add support for power it without a USB connection.

Well, as an example the mask roms (kernal, basic, char roms) in a C64 use 30-90 mA (avg ~65 mA) - so I think that you are good to go  :-)

But yes - an option to pull down reset while copying from flash to SRAM would be nice to have. Also multiple units should be able to join in on the same reset line...

Thanks :-)

wickerwaka, 2024-12-13:
#13 adds support for powering from vsys

closed 2024-12-13 by wickerwaka

UffeJakobsen, 2024-12-13:
> #13 adds support for powering from vsys

Nice - thanks :+1: