wickerwaka/PicoROM #10: No longer compiling with current sdk

sbrk-r, 2024-12-06:
Hi, great project!

It seems the library function, 'set_sys_clock_khz()' has been moved into "hardware/clocks.h"

After making that change I was able to compile and it seems to work - I don't have real hardware yet, but a dev board enumerates and communicates with the picorom host software.

PS - any reason to not have a fast OR or NOR gate summing CE# and OE# to free up a pin..? I think I'd like to change the IO expander to something a bit more traditional and visible to the naked eye.

wickerwaka, 2024-12-06:
Thanks. I have another branch that I'm doing some cleanup in and already have this fix there, I will merge it back into main soon.

As far as the /CE and /OE pins are concerned. Yeah, it would be nice to use an I2C IO expander instead of the pretty obscure TI one-wire part. Also having the /CE and /OE inputs combined externally would make some of the PIO code a lot simpler because it could just wait or jump based on a single pin value.

The reason I haven't done this is because I want to maintain compatibility with as many ROM pinouts as possible. Someone recently needed to use this to replace a non-JEDEC 27C1000 which has A16 and /OE swapped. I was able to make some changes to the firmware to support that, which I wouldn't have been able to do if that signal was being combined externally. Will probably never be needed again, but I want to maintain as much flexibility as possible.

I've also considered using Dual SPI instead of Quad SPI for the flash since flash access speed isn't critical. That would free up two pins that could be used to bit-bang I2C from the CPU.

sbrk-r, 2024-12-10:
That makes sense. Dropping to dual spi is interesting, I hadn't realized those pins could be repurposed.

sbrk-r, 2025-03-11:
Reopening because I can't figure out how to talk to you otherwise, sorry if that is annoying (!)

My immediate use for this is 28pin EPROMs, probably would have been smarter to just build an adapter but I kludged a board together instead... 
[image]

Very few hardware changes, added an OR gate summing OE# and CE# and putting them in the old OE# pin (20). old CE# pin (21) and TCA IO expander pin (18) are routed to a PCA8574. Bit easier to solder by hand.

Unfortunately I read the datasheet _after_ I submitted it to the board house, and you need sequential pins for the HW I2C (also stipulated in the PIO I2C example). So I'll have to figure out something in software, and learn more about the PIO.
 
Strangely, though - it works with the stock firmware (ignoring expander functions). Since CE# is now used for I2C and pulled up with a resistor, it should never trigger the !OE && !CE logic, unless i misinterpreted that... maybe there is some bug there. or my board has a short.

sbrk-r, 2025-07-15:
(back from vacation now) Yes! I meant to update. I forgot I had jumpers that defaulted to the same pinout of OE# and CE# as your board, so it could be used with either IO expander. I also inadvertently left the LEDs as grounded cathode with won't work with an open collector driver (PCA8574), but with one wire mod I did get it to work without issues... other than writing the software which took a bit to get my head around RP2040.

It's not too elegant but the changes are in my fork at https://github.com/sbrk-r/pikorom if you were curious.

closed 2026-02-13 by wickerwaka