wickerwaka/PicoROM #11: Too slow for intel Pentium 100 motherboard ?

Deksor, 2024-12-09:
Hello !
This project is going to be really helpful for me to debug an AT 286 motherboard that has some issues and LOTS of components to test (it's mostly made of glue logic).

### Problem
I just built one myself, and during testing, while it seems like it's doing ok in a 386 motherboard, it's struggling to get a Pentium board to POST.

What I did is that I took out the original ROM (which contains an award BIOS v4.51PG), dumped it, then flashed the bin file to the picorom while making sure of using the right size (1MBit).
Then I put a POST card into an ISA slot of the motherboard to monitor the activity and finally I turned it on.

The board executes some code, but after a while hangs.
When I put back the original ROM chip in place of the picorom, the board POSTs fine to the memory count screen, so the motherboard works fine.

Here's a video of the picorom in action https://youtu.be/1Ub2xlTqZPE

### Informations / context
Then I pulled off the sticker of the ROM chip and it's an Atmel AT29C010A-**15PC** AKA 150ns ROM according to the datasheet, yet the picorom is said to be able to run at 100ns.

If you need informations about the motherboard, it's a PcPartner TXB820DS https://theretroweb.com/motherboards/s/pcpartner-txb820ds-35-8333-00,35-8333-01,35
CPU is a Pentium 100

Just for your information if you've never used a POST card, the relevant pair of hex numbers is the one at the top. It starts with C1 on an award BIOS which is normal, then does some progress and finally hangs at 0x05 which is according to here https://blog.theretroweb.com/2024/01/20/award-bios-beep-and-post-codes-list/ after "keyboard initialization". Referring to the "version 4.0" it might be hanging during the next code 0x06 aka "ROM checksum" (though 0x06 is missing from the listing for version 4.51PG I think it's still where it hangs ...)

I also tried the picorom in a TL866 and the file it outputed was exactly identical to the original one, so I think my unit is working just as expected.

I used this version of the firmware https://github.com/wickerwaka/PicoROM/releases/download/v1.5/PicoROM.uf2 on the pico.

### Conclusion
Can you help me ?
I can provide any other piece of information needed for the troubleshoot

wickerwaka, 2024-12-10:
It's possible it's still too slow, I need to go back and run some speed tests on the 1.5 version with a new benchmarking version I have setup.

With the latest version I am overclocking the CPU more and the access time is 70ns. It's not merged into main, try this firmware here: https://github.com/wickerwaka/PicoROM/actions/runs/12249629253/artifacts/2297819714

Deksor, 2024-12-10:
Thanks a lot for the new firmware.

I tried it and now the board posts !

However multiple times it also got stuck at code C0 which is the very first one with that bios afaik.

I didn't have the time to experiment more so, in this case, maybe it's just a hiccup from the motherboard, I'll see tonight if I can try it more.

closed 2024-12-10 by Deksor

wickerwaka, 2024-12-10:
At this speed the access time should be more than enough to match the 150ns of the original ROM. I wonder if I'm not hitting some of the other timing expectations.

[image]

The "/CE or /OE to Output Float" time at 40ns might be right on the border. I made a slight firmware change that should turn off the buffer about 8ns faster. Maybe it'll fix it, but probably not. 

https://github.com/wickerwaka/PicoROM/actions/runs/12265755534/artifacts/2302710479

Let me know.

Deksor, 2024-12-10:
This is much better.
It still crashes sometimes early in the POST, but it seems to be related to the power on cycle, not sure exactly why. I couldn't get it to crash later, during POST or anywhere.

wickerwaka, 2024-12-10:
Okay good. I'm going to try to build some tests to measure these other values.

wickerwaka, 2025-01-27:
If you are still experimenting with this, please try the latest firmware (version 1.7).

Deksor, 2025-01-28:
Yes I'm still experimenting !

You did a great job !

No more crash early during POST, however the board seems to have developped a fault with the keyboard, it's really strange. I'm going to investigate more with another motherboard, I'm not sure if the cause is actual hardware or something else.

I've also had an error I've never seen after it POSTed where it displayed "Unknown flash", but I assue this is because the motherboard is trying to flash the PNP table or something like this (I can't remember what this is called exactly but TLDR the board tries to write something to the rom but doesn't recognise its type ; actually maybe adding a fake ROM identifier that would be recognized by most hardware could be added ?). Also I'm not sure if as-is the picorom can be written to by the destination board.
I know you said it can be used to communicate between host and destination board through the picorom in a previous version of readme, but that's an entirely different thing (very nice feature though, that will be handy in my upcoming projects 😄 )

Anyways, let me investigate more on different motherboards, but so far it seems promising !