wickerwaka/PicoROM #16: Linux picorom executable doesn't recognise the picoroms connected to the PC, but my compiled executable does

Deksor, 2025-01-28:
### Problem

I downloaded the linux version of the picorom tool and tried to execute it. The tool runs, but for some reason it doesn't recognise my pico.
I can see the "link" led flash on my picos when I run that command, but yeah they're not being recognized as the program returns "No PicoROMs found."

Meanwhile I compiled the repo's code and it detects them fine.

### Informations / context

My system is based on ryzen 5000 series with x570 chipset and I run Debian 12. My user is part of the dialout group in order to have access to the pico.

wickerwaka, 2025-01-28:
Can you confirm that you are using the 1.7 version of the firmware and the picorom tool?

Deksor, 2025-01-28:
Yes ! I downloaded the new host program right after I flashed my pico

wickerwaka, 2025-01-28:
Thanks. I'll see if I can reproduce it tonight.

wickerwaka, 2025-01-29:
Fixed by 21f895213d67d2e4af0787dfe27bbebee17e2007

closed 2025-01-29 by wickerwaka